andrewhodel/grblweb #18: Preview Window

danst0, 2015-10-21:
Hi,

I have not been able yet to produce one GCode file that shows up in the preview. In the Result window I regularly receive the following:

--------------New Gcode Loaded--------------
Min Dimensions X: 100000 Y: 100000 Z: 100000
Max Dimensions X: -100000 Y: -100000 Z: -100000

Total Dimensions X: -200000 Y: -200000 Z: -200000

Which are definitely not the dimensions of my GCode. I would be happy to supply sample code.

Daniel

andrewhodel, 2015-10-21:
Upload or paste the gcode.

Also paste the output of the browser console when you open a gcode file.

danst0, 2015-10-21:
(Projekt 20151020 Fraeserablage_v2)
(Erstellt mit Estlcam Version 8 Build 8,047)
(Laufzeit ca. 00:28:12 Stunden)

(Benoetigte Werkzeuge:)
(Soro 2.4mm L2S)
(China 6mm)
G90
M03 S24000
G00 Z5.0000

(Nr. 1: Tasche raeumen: Ausschnitt 1)
G00 X4.5000 Y80.5600
G00 Z0.5000
G01 Z-1.0000 F500 S24000
G01 Y1.4400 F600
G00 Z5.0000
G00 Y80.5600
G00 Z-0.5000
G01 Z-2.0000 F500
G01 Y1.4400 F600
G00 Z5.0000
G00 Y80.5600
G00 Z-1.5000
G01 Z-3.0000 F500
G01 Y1.4400 F600
G00 Z5.0000
G00 Y80.5600
G00 Z-2.5000
G01 Z-4.0000 F500
G01 Y1.4400 F600
G00 Z5.0000
G00 Y80.5600
G00 Z-3.5000
G01 Z-5.0000 F500
G01 Y1.4400 F600
G00 Z5.0000
G00 Y80.5600
G00 Z-4.5000
G01 Z-6.0000 F500
G01 Y1.4400 F600
G00 Z5.0000

(Nr. 2: Tasche Seitenwand: Ausschnitt 1)
G00 X4.2000 Y1.2000
G00 Z0.5000
G01 Z-1.0000 F500
G01 Y80.8000 F600
G01 X4.8000
G01 Y1.2000
G01 X4.2000
G01 Z-2.0000 F500
G01 Y80.8000 F600
G01 X4.8000
G01 Y1.2000
G01 X4.2000
G01 Z-3.0000 F500
G01 Y80.8000 F600
G01 X4.8000
G01 Y1.2000
G01 X4.2000
G01 Z-4.0000 F500
G01 Y80.8000 F600
G01 X4.8000
G01 Y1.2000
G01 X4.2000
G01 Z-5.0000 F500
G01 Y80.8000 F600
G01 X4.8000
G01 Y1.2000
[... 4747 lines cut ...]
G03 X52.6327 Y101.4383 I-3.0000 J0.0000
G03 X50.2122 Y102.9925 I-2.6327 J-1.4383
G01 X27.3073 Y103.0000
G00 Z-16.0000
G01 X20.1073
G01 Z-18.0000 F300
G01 X0.0000 F500
G01 X-0.2995 Y102.9850
G03 X-2.6736 Y101.3608 I0.2995 J-2.9850
G03 X-3.0000 Y99.9800 I2.6751 J-1.3613
G01 Y54.3610
G00 Z-16.0000
G01 Y47.1610
G01 Z-18.0000 F300
G01 Y0.0000 F500
G03 X-2.7632 Y-1.1683 I3.0000 J0.0000
G03 X-0.2122 Y-2.9925 I2.7632 J1.1683
G01 X19.1381 Y-3.0000
G00 Z-16.0000
G01 X26.3381
G00 Z5.0000
G00 X0.0000 Y0.0000

M05
M30

andrewhodel, 2015-10-22:
The problem here is that the gcode viewer doesn't expect padded Gcodes, so it's looking for G1 not G01.

I will add that so that is resolved.

Another issue is that there are arcs (G2 and G3), I suppose I could add support to display them however for my workflow with Millcrum both the design and tool options are set in the same file so there is no benefit to having the machine draw out the arc line segments.

I can see however, that if you were using a normal DXF->CAM->GCODE process that you might want the machine to handle the arc -> line segments because it would "know" the bit dimensions and could properly draw the line segments.  However as I build this stuff mostly for myself (I don't get paid anything) I just build for what I need.

For the moment I'll get the 00 01 padding fixed and you are more than welcome to submit a patch to add arc support.  If it will help you out, here's where the code would go - https://github.com/andrewhodel/grblweb/blob/master/i/gcode-viewer/gcode-model.js#L95 and here is some javascript code to process an arc into a line segment (this is for DXF arcs but you could get the idea) - https://github.com/andrewhodel/millcrum/blob/master/inc/dxf.js#L201

I'll get the padding done once I get OpenBSD working with the viewer (that's what I'm working on right now).  It has a problem with WebGL.

closed 2020-09-01 by andrewhodel