andrewhodel/grblweb #20: npm install fails

Bradan, 2016-02-08:
Hello,

I tried to execute `npm install` and it fails:

    $ npm install           
    grblweb@ /home/username/grblweb
    `-- (empty)

    npm WARN EPACKAGEJSON grblweb@ No repository field.
    npm WARN EPACKAGEJSON grblweb@ No license field.
    npm WARN EBUNDLEOVERRIDE Replacing bundled serialport > node-pre-gyp with new installed version
    npm ERR! Linux 4.1.15-3-ARCH
    npm ERR! argv "/home/username/.nvm/versions/node/v5.5.0/bin/node" "/home/username/.nvm/versions/node/v5.5.0/bin/npm" "install"
    npm ERR! node v5.5.0
    npm ERR! npm  v3.3.12
    npm ERR! path /home/username/grblweb/node_modules/serialport/node_modules/node-pre-gyp/node_modules/mkdirp/bin/cmd.js
    npm ERR! code ENOENT
    npm ERR! errno -2
    npm ERR! syscall chmod

    npm ERR! enoent ENOENT: no such file or directory, chmod '/home/username/grblweb/node_modules/serialport/node_modules/node-pre-gyp/node_modules/mkdirp/bin/cmd.js'
    npm ERR! enoent This is most likely not a problem with npm itself
    npm ERR! enoent and is related to npm not being able to find a file.
    npm ERR! enoent 

    npm ERR! Please include the following file with any support request:
    npm ERR!     /home/username/grblweb/npm-debug.log
    npm ERR! code 1

Could you help me with this? I tried installing the node-pre-gyp, but that doesn't help.

[npm-debug.log.txt](https://github.com/andrewhodel/grblweb/files/121708/npm-debug.log.txt)

andrewhodel, 2016-02-08:
Try this:

```
rm -rf /home/username/grblweb/node_modules
cd /home/username/grblweb
npm install
```

andrewhodel, 2016-02-08:
And if that doesn't work it could be that you are using node 5.5.0 which is a very old version.

If you have further issues take it to the support mailing list as this is not a bug it is a user issue.

closed 2016-02-08 by andrewhodel