andrewhodel/grblweb #23: grbl v1.1 support

michelcve, 2017-01-13:
Any chance that grblWeb will support the new grbl v1.1 protocol soon?

hrobeers, 2018-01-10:
@michelcve Is it the new jogging commands you are aiming at? Or are there other protocol features you miss?

michelcve, 2018-01-10:
Wow, that comment was a year old, so I actually forgot about it ;-)

I've since moved to a different (commercial) product for my CNC, so I've kind of lost interest in grblWeb, sorry.

Protoneer, 2018-01-10:
What did you move to?

hrobeers, 2018-01-10:
@michelcve I actually just discovered grblweb and chose is because it does not require a screen attached to my PC that talks to the controller.
As a software developer I can modify this code to my liking, so I'm wondering how grblweb turned you off.

michelcve, 2018-01-11:
I switched to [EstlCam](http://www.estlcam.com). It's very reasonably priced, and does everything I need it to do.

Being a software developer, I understand that having the source code and being able to modify things is nice. But in the end, Estlcam simply does everything I want. And ultimately I started CNCing for the CNCing, not the programming (enough other projects to satisfy my programming needs anyway ;-)).

Hoangthoi, 2018-07-10:
[image]