andrewhodel/grblweb #28: Raspbery Freeze

qiwalker, 2019-01-31:
hi can someone help me, my raspberry getting freeze after 5 minute running using grbl web :(
harware
Raspberry pi 3 B+
raspian strech
grbl 1.1e

thanks
Rizki

andrewhodel, 2019-01-31:
Can you ping it when it is frozen?

Andrew

qiwalker, 2019-01-31:
Dear Mr andrewhodel

no sir, need to restart

andrewhodel, 2019-01-31:
And what does dmesg say if you are tailing it when the crash happens?

What does the node log say?

You can log both of these within ssh or in a tty.

Andrew

qiwalker, 2019-02-07:
Hi Mr @andrewhodel 

apologize for late reply
node log say
"File server error: {status: 404}"
and node static with message : not found
just that

kindly please update for support raspberry pi3 b+ sir

thanks
Rizki

andrewhodel, 2019-02-07:
That is not what I asked for.

Was there a term or word you were unclear of the definition of?

If so we can go into those in order to resolve the problem, as you do want
it to work correct?

Andrew

qiwalker, 2019-02-14:
Dear Mr. @andrewhodel 

[image]

i got error in Mpos and Wpos using grbl 1.1 as above picture

thanks
Rizki

andrewhodel, 2019-02-14:
https://groups.google.com/forum/m/#!topic/xyzbots-support/

This issue has already been raised, I pointed out the new protocol GRBL
uses and exactly where in the GRBLWeb code you can write support for it.

Andrew

BaxterZZZ, 2020-08-29:
I looked at the google groups post and others, and I fixed parsing of the "?" command's reply. However, It still crashes on my RPi 3b with this erro sequence:
pi@rpi3:~/grblweb $ sudo node server.js
Got error: connect ECONNREFUSED 127.0.0.1:8080 not enabling webcam
connected at 115200 /dev/ttyAMA0
connected at 115200 /dev/ttyUSB0
user wants to use port 1
switching from undefined

<--- Last few GCs --->

[5455:0x2ace288]   158293 ms: Scavenge 229.7 (231.4) -> 229.3 (231.4) MB, 6.8 / 0.0 ms  (average mu = 0.241, current mu = 0.264) allocation failure 
[5455:0x2ace288]   158333 ms: Scavenge 229.7 (231.4) -> 229.4 (231.4) MB, 7.1 / 0.0 ms  (average mu = 0.241, current mu = 0.264) allocation failure 
[5455:0x2ace288]   158374 ms: Scavenge 229.8 (231.4) -> 229.4 (232.4) MB, 7.1 / 0.0 ms  (average mu = 0.241, current mu = 0.264) allocation failure 

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
Aborted
pi@rpi3:~/grblweb $ 
Can anyone help me fix this?  BTW, I already tried updating to node v12.x No help there.

andrewhodel, 2020-09-01:
There is another mailing list thread a similar problem with the node env.

Andrew

closed 2020-09-01 by andrewhodel

BaxterZZZ, 2020-09-01:
Andrew, could you please link to this thread you're suggesting that I look at?
Thank you!
PS: I took all of the code that opens the USB, sends "?" and parses the reply into a separate program, requiring only 'serialport'. It had no problem running indefinitely using only 0.3% cpu and 3% memory. Please help me find where the memory is getting used up (?)

andrewhodel, 2020-09-01:
It is something with the heap size of node.

Just google xyzbots mailing list then search heap after you click the link.

Andrew

BaxterZZZ, 2020-09-01:
I don't know if this link works for you...
https://groups.google.com/forum/#!searchin/xyzbots-support/heap|sort:date
But the results is:
 Results for heap in XYZBots Support for GRBLWeb and RepRapWeb Search all groups
Sort by relevanceSorted by date
	
Results: about 0 for heap
No results found

andrewhodel, 2020-09-01:
Yea somewhere in there the thread will match your error enough.

Andrew

BaxterZZZ, 2020-09-01:
Andrew,
1) I don't see how any topics can match my error if none of them contain the word "heap" which is in my error msg twice. 2) I looked through all 63 topics and see nothing related to this subject. Is it possible the post you are thinking about was deleted or located somewhere else? Since you seem to know something about this, I'd really appreciate a link that you find right now that is on-topic.  Thanks!!!

andrewhodel, 2020-09-01:
Must be somewhere else, it is a heap problem though iirc.

andrewhodel, 2020-09-01:
Another entry point that would result in the same resolution is
experimenting with the node command line options.

There are options for heap size.

andrewhodel, 2020-09-01:
You gotta understand the real thing is I don’t have that exact model device
sitting on a bench in the shop I don’t have just waiting to be tested.

I could ssh to yours, it is just somewhat frustrating like changing your
friends oil instead of letting him use the lift.

Get what I’m saying? I don’t get paid for this stuff.

andrewhodel, 2020-09-01:
And sometimes that oil change can result in a transmission rebuild!

BaxterZZZ, 2020-09-01:
The only reason you might have to fix the transmission is if it has a leak. So are you saying that your latest release should run indefinitely if the heap size is increased a bit... My RPi has 1Gb ram, and by default node gets 512Mb, which seems like a lot for handling a little web and USB traffic. For example, I could increase the memory to 750Mb with the command node --max-old-space-size=750 server.js, or could it be that your code as a leak where you are not releasing memory so it can't be garbage collected and the program will blow up eventually no matter what?  I realize that no one is paying you to maintain your code, but hopefully you like that it lives on and take pride in people being able to use it years after you wrote it :)  Any help in solving this is appreciated.

andrewhodel, 2020-09-01:
You don’t seem to understand making a transmission then.

andrewhodel, 2020-09-01:
Also you can run the code on node compiled for x86_64 and watch the memory
usage, there is no leak in the grblweb code.

Andrew

BaxterZZZ, 2020-09-02:
i just added the line "console.log(process.memoryUsage());" to server.js at the end of serialData on the RPi and the usage just keeps increasing :/ 
{
  rss: 59822080,
  heapTotal: 28471296,
  heapUsed: 23715004,
  external: 1367010
}
.......
{
  rss: 250036224,
  heapTotal: 213020672,
  heapUsed: 207757256,
  external: 1391627
}
...etc.
Any idea what the difference could be between x86 and ARM platforms? I think that you originally had a disk image for the RPi (which I haven't used because the code needs changing for my purposes). Something is blowing up even with a fresh install from ur github.

andrewhodel, 2020-09-02:
As I tried to tell you earlier, I don’t have the hardware just sitting
here.

BaxterZZZ, 2020-09-02:
I wrote to you privately to see if you will take a look using my hardware via ssh. It would be great to get this working!

fleshmanj, 2021-12-31:
@BaxterZZZ did you ever get this working. I am also having this same issue and it doesnt matter how much memory i give node my programs keep failing halfway through. Very frustrating......

BaxterZZZ, 2022-01-01:
@fleshmanj No I didn't. I spoke with Andrew Hodel, and he kept trying to convince me that it was due to incompatibilities between different more up-to-date versions of the libraries his program uses, but I used exactly the ones referenced in his manifest and the memory leak was identical. He didn't seem to have any interest in helping or to keep the software package viable for others, so, as a nodejs novice, I stopped trying to use it.  Seems like a shame, because it had several features that were exactly what I wanted for a different project: to use the 3-axis grbl machine to move a camera from place to place. But that would be for a long term observation, so having software that reliably crashes on the RPi was not acceptable. I tried rewriting the server part using more modern libraries like express, but got in over my head. Maybe he will be in a better mood after seeing your post, if he sees it, and offer some help this year.... If you figure this out, please let me know!!!

fleshmanj, 2022-01-01:
@BaxterZZZ after reading through some of the comments and decided to try a different route. So far I have had zero issues and now i can use my phone to operate the XCARVE i have. If you want some help I could send you some links. I basically did what this guy did but i run mine headless and use VNC to my laptop if i need to do anything graphically. https://millrightcnc.proboards.com/thread/2207/setup-raspberry-pi-ugs. 

Hope you get your setup working. I can tell you I love not having a $2K laptop getting covered in dust.

Cheers!