andrewhodel/grblweb #29: Grblweb on CNC laser engraver

JalenTk, 2019-03-03:
Is it possible to use Grblweb to control a CNC laser engraver?

andrewhodel, 2019-03-03:
Yes, it will send whatever is in the gcode but it just won't show the
on/off of the laser in the viewer.

You'd just need to add support for it, it's just basically sending gcode
data to a serial port but does some nice things like pause/play/restart and
showing the gcode as a visual model.

You opened and closed it as I was 80% through this message but already
typed the first sentence :P

Andrew

JalenTk, 2019-03-03:
Sorry @andrewhodel , I've misclick the close button XD

Another question, is it possible to access to GRBL web from another network, I mean like from out of home and using mobile data when my CNC is sitting at home?

Thanks in advanced!

andrewhodel, 2019-03-03:
You have to know TCP/IP, but of course.

Andrew

JalenTk, 2019-03-03:
Good to know!
Thanks!

closed 2019-03-03 by JalenTk