andrewhodel/grblweb #30: Interface look different, without the X,Y,Z control

JalenTk, 2019-03-08:
[image]

Why is my interface looks different with yours? It doesnt have the X,Y and Z control.

Also, after sending Gcode to CNC, there is no response, coming with a fileServer error. What can I do with it?

Thank you in advanced.

andrewhodel, 2019-03-08:
What is my interface?

The interface you are using looks just like the one from the latest code.

If you want the previous interface, like in some of the older Youtube
videos, use a commit prior to
https://github.com/andrewhodel/grblweb/commit/208f83eed41ae60f531eda5ec191040452901b86

I don't know what, "Also, after sending Gcode to CNC, there is no response,
coming with a fileServer error. What can I do with it?" is trying to say.

If you have an error, you can try to use the the error to resolve the
issue.  If you don't tell anyone the error though, nobody can really
explain to you how to fix it.

Andrew

andrewhodel, 2019-03-08:
Like my typing the the instead of the, that's a typo but is also an error.

We could just leave it as the the and expect that everyone who has typed
that error would be in the know, but that would get ridiculous.

English is pretty adequate at explaining this stuff, some of the Spanish
conjugations are nice but most development is done in English.

Understand that I could say all of that, or just edit it.  With computer
code and Git we use commits because it is a lot easier to keep track of
things.

A support request just doesn't need commits though.

Important concept to understand.

So what's your fileserver error, assuming you understand the interface you
are using and where different commits are?

I agree that versions would be better for this particular situation, but
it's not enough to motivate me to make versions on unpaid work that at the
time was updated very often and released throughout it's development.  It
is also a platform that anyone can modify if they have the skills,.

What I am saying is, I'll try to help you with the error you are getting
but the software works.

JalenTk, 2019-03-13:
`pi@raspberrypi:~/grblweb $ sudo node server.js
Got error: connect ECONNREFUSED 127.0.0.1:8080 not enabling webcam
connected at 115200 /dev/ttyAMA0
connected at 115200 /dev/ttyACM0
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
user wants to use port 1
switching from undefined
`
For the fileServer error, does that mean that I have to install that particular release of node-static in order to solve it? 

Thanks for replying sir.

andrewhodel, 2019-03-13:
Why not just download a prebuilt image and see what version it is running?

Andrew

andrewhodel, 2019-03-13:
You can write the .img file to a block device after decompressing it like
this.

dd if=xyzbots-reprapweb-grblweb-2015-may.img of=/dev/sdc bs=4M

You can make a block device out of a file like this -
https://www.jamescoyle.net/how-to/2096-use-a-file-as-a-linux-block-device

Andrew

andrewhodel, 2019-03-13:
Also, it seems like it's working.  Those errors might be because a file
like favicon.ico don't exist or some other client is requesting an unknown
path.

For example, the road has an error from the planning department if there
isn't a road to a particular segment of the city.  In this same way, the
server notifies.

Understand?

Andrew

JalenTk, 2019-03-13:
Alright Andrew, I'm trying to work on it!

Thanks for replying sir!

closed 2020-09-01 by andrewhodel