andrewhodel/grblweb #32: Raspberry PI 4 -- FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

tseanard, 2020-06-08:
Hello,

I am trying to use GrblWeb on a Raspberry Pi 4b 4GB, and I am having issues.  The following is the console output on the terminal.  The key issue is the JavaScript heap out of memory failure that crashes the app.  Once this happens, I have to re-home in order to proceed, and it happens after just a few minutes.  I have tried with Ubuntu 18.04, 19.10, 20.04. Raspberry PI OS 64 bit and Raspberry Pi OS 32 bit, all with the same issue.  Current node.js version is v10.21.0, but I have also tried it with v12 and v14 with the same issue.  For troubleshooting, I disabled the '?' polling because i thought that the hits on the sockets.emit may be the memory leak, but that also did not resolve the issue.  I do have the camera, and had set up the mjpg stream (which was working), but was still having this issue below even with the camera set up and operating properly.

...
pi@raspberrypi:~/grblweb $ sudo node server.js
Got error: connect ECONNREFUSED 127.0.0.1:8080 not enabling webcam
connected at 115200 /dev/ttyAMA0
connected at 115200 /dev/ttyS0
connected at 115200 /dev/ttyUSB0
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
user wants to use port 0
switching from undefined
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
user wants to use port 0
switching from undefined
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }
fileServer error:  { status: 404,
  headers: { server: 'node-static/0.7.9' },
  message: 'Not Found' }

<--- Last few GCs --->

[1386:0x717c18]   333339 ms: Scavenge 695.8 (724.6) -> 694.3 (724.6) MB, 8.1 / 0.0 ms  (average mu = 0.186, current mu = 0.170) allocation failure
[1386:0x717c18]   333436 ms: Scavenge 696.0 (724.6) -> 694.5 (724.6) MB, 8.1 / 0.0 ms  (average mu = 0.186, current mu = 0.170) allocation failure
[1386:0x717c18]   333533 ms: Scavenge 696.2 (724.6) -> 694.7 (728.6) MB, 8.1 / 0.0 ms  (average mu = 0.186, current mu = 0.170) allocation failure

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
Aborted
pi@raspberrypi:~/grblweb $
...

andrewhodel, 2020-06-08:
My code runs fine here, I don't know what's wrong but if you want to work
on it I have absolutely no financial interest and don't care at the moment.

Sorry,
Andrew

tseanard, 2020-06-17:
Thank you for confirming that this code does still work somewhere.  It appears that the real root of this issue is related to incompatibility between things (packages and OS) that I have installed on my system, not with this project.

closed 2020-06-17 by tseanard

andrewhodel, 2020-06-18:
You can build nodejs from source on the pi. 

Thank You,
Andrew Hodel