andrewhodel/grblweb #33: Got error: connect ECONNREFUSED 127.0.0.1:8080 not enabling webcam

ProgrammerZ001, 2020-08-20:
Using node js on window 10 
trying to run grblweb

config.webPort = 85;
config.serialBaudRate = 115200;
config.usettyAMA0 = 1;
// expects a webcam stream from mjpg_streamer
config.webcamPort = 8080;

my configurations

Got error: connect ECONNREFUSED 127.0.0.1:8080 not enabling webcam
[image]

andrewhodel, 2020-08-20:
Did you read the part about mjpg_streamer?

You would have to port that to Windows to do what you want.

ProgrammerZ001, 2020-08-20:
Thank you so much, Andrew, for such a quick response.
can please guide me a little bit
I am using mjpeg streamer by flyingorilla from the window store, will it do
the job, what I understand from reading is that I have to stream the USB
cam because node can get to do so. Am I right?
 thank and regards
Muhammad Zubair
software engineering student final year

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
08/20/20,
11:47:48 PM

andrewhodel, 2020-08-20:
What are you trying to figure out?

Capture the MJPEG stream from mjpg_streamer running on linux using tcpdump
and then find a program that recreates the exact same protocol (http) and
file type (mjpeg) on windows.

Then point grblweb to that url.

Andrew

andrewhodel, 2020-08-20:
This is why many people think Linux to be more useful than Windows.

Andrew

closed 2020-09-01 by andrewhodel