andrewhodel/grblweb #35: Please add hardware requirements to the README.md

DanielYWoo, 2020-12-11:
There are multiple raspberry pi versions in the market, and some compatible hardwares like orange pi. Could you add some minimum and recommended hardware requirement to run grblweb?

e.g, Raspberry 1/2/3/4, RPI Zero, RPI Zero W, Orange PI, etc.
They have different versions with different memory, does it work with 512MB or at least 1G?

andrewhodel, 2020-12-11:
It will run on architectures and hardware other than the pi also.

I bet you could get it running on sparc with 128mb of memory if you had a
(whole) day to compile node.

DanielYWoo, 2020-12-12:
Hi Andrew,

I know nodejs can run on different platforms, my question is what kind of
hardware to run it smoothly. I bet Pi Zero will be very very slow to run
it. ;-)

Thanks,
Daniel

andrewhodel, 2020-12-12:
The slicer may be slow, but for many people they save the gcode files and
remake objects.

The sending of gcode to the control board is serial anyway, that’s been
around for a long time. You could do that on an atari.

DanielYWoo, 2020-12-13:
I generate gcode on my Mac/PC, then upload to Pi. So, my Pi needs WIFI, a web server (CNCjs) and a web cam, maybe a PSU relay control, plus a smoke sensor to stop the milling if it's on fire.
I have a Pi 2B with my Marlin based 3d printer, the 4-core CPU can do 7000 million ops/second, in most cases it works fine, but some plugins could slow down Pi then I see Pi is sending gcode to my printer with obvious delay, the nozzle moves and pauses. So I have to keep my Pi doing nothing when printing, no uploading, no config changes.
I kind of hope CNCjs is more performant than Octoprint.

closed 2020-12-30 by DanielYWoo

andrewhodel, 2020-12-30:
With the recent changes described in the mailing list, it works on all pi
hardware and anything that runs node.

Andrew

DanielYWoo, 2021-01-01:
that's awesome, I think if we have only web interface, no heavy CPU work like arc-welder in OctoPrint, even Pi Zero should work smoothly. 
And, sorry I was confused with gbrl-web and cncjs, both are cool anyway.